siemens/embb #7: mtapi_c Fibonacci crushes

sunway513, 2015-02-23:
Hello,
We have tested the Fibonacci example code on our server. And we found the code will cause "core dumped" once the parameter is larger than 15; the program runs normally with parameter of 15 and smaller. 
The server has Intel(R) Xeon(R) CPU E5-2640, 64GB memory running with Fedora 18. 
Do you have any suggestions?
Thanks!

Regards,
Peng

tobias-schuele, 2015-02-24:
Hi Peng,

to avoid dynamic memory allocation, which is often forbidden in embedded systems, the number of active tasks EMB² (MTAPI) can handle is limited. If more tasks are created, an exception is thrown or, in case exceptions are disabled during compilation, the program aborts. By default, the maximum number of tasks is 1024. You can change (increase) this value by explicitly initializing the task scheduler using embb::mtapi::Node::Initialize or the corresponding C function.

Note that a recursive computation of the Fibonacci numbers creates a large number of tasks: To compute Fib(n), there are 2*Fib(n+1)-1 calls to Fib. For n=15, this means that 1973 tasks are created, which exceeds the default limit. As mentioned previously, however, the program should abort normally instead of creating a core dump. We will try to reproduce the problem.

Regards
Tobias

marcus-winter, 2015-02-24:
Hi,

the Fibonacci example uses the MTAPI c interface, so there are no exceptions. The MTAPI_CHECK_STATUS macro calls abort() in case of an error which in turn yields the message "Aborted (core dumped)".
Concerning the maximum number of tasks please note that the default limit is 1024.

Best regards,
Marcus

closed 2015-02-24 by tobias-schuele

marcus-winter, 2015-02-24:
The MTAPI_CHECK_STATUS macro has been changed to emit an error message and call exit() instead of abort().