siemens/embb #13: Zip release files

tobias-schuele, 2016-11-11:
Currently, only a .tar.gz file is generated for releases. This is a bit cumbersome for Windows. Additionally, a .zip file would be useful.

christian-kern, 2016-11-11:
The create_tarball script has to be adjusted, to additionally generate a zip file. Currently, a tar.gz is created this way:

```
#finally, build the tarball.
echo "--> Calling tar"
tar -czf $TARBALL_NAME -C $MYTMPDIR ${n}
```

Afterwards, Jenkins has to be adjusted, so this ".zip" artifact is kept, and can be downloaded by the user, like it was done for the tar.gz.

christian-kern, 2016-11-11:
See https://github.com/siemens/embb/commit/74ef64d633bf1e6ecc7d6b865387ee314659a1a3

Jenkins is not set up yet as the branch is not yet merged to master.

tobias-schuele, 2016-12-13:
Both the .tar.gz and the .zip file contain a file "myfile.properties" in the root directory which should be removed.

marcus-winter, 2016-12-16:
Verified on Jenkins, closing.

PR #27 (2016-11-23): Zip release files
Fixes #13
closed 2016-12-16 by marcus-winter