siemens/embb #15: Add mutex implementation for atomic operations to support threading analysis tools

tobias-schuele, 2016-11-11:
As a user of EMB², I want to be able to run dynamic threading analysis tools such as Helgrind or ThreadSanitier on my application. However, most tools are not able to deal with atomic operations, especially from external libraries. To avoid false positive results, atomic operations should be "simulated" using mutexes for the analysis.

Add an implementation of atomic operations at C level using EMB² mutexes. For example, when calling embb_base_atomic_compare_and_swap_int, the CAS should be protected by a mutex (embb_mutex_t) without any assembly code. The implementation is enabled by a CMake option which should only be used for analysis.

tobias-schuele, 2016-11-11:
Comments from JIRA:

See https://github.com/siemens/embb/commit/3926d7a1b7ba4f1f51290ea8b949b7957ad019e0 .. https://github.com/siemens/embb/commit/a7c7b027cbd1ceea9285a7422f208a917d2a2f8a

--------------------------------------------------

Describe CMake option in new README (section on detailed installation instructions)
Create comments in source code for static atomics / spinlocks to indicate that potential warnings are false positives and the implementation is intended.
Add Doxygen attributes concerning thread-safety / progress guarantees to embb_atomic_init_TYPE and embb_atomic_destroy_TYPE
base_c/include/embb/base/c/internal/atomic/and_assign.h: Cut-and-paste error?
	* following generator macro is used. This macro is called by the macros in the
+ * following generator macro is used. This macro is calleuild-work-Desktop-Defaultd by the macros in the
Why are EMBB types / functions replaced with corresponding internal ones? Example:
	int compare_to = 0;
	if (embb_atomic_load_int(&embb_thread_index_counter_flag) != 2) {
	if (embb_atomic_compare_and_swap_int(
+ EMBB_BASE_BASIC_TYPE_SIZE_4 compare_to = 0;
+ if (embb_internal__atomic_load_4(&embb_thread_index_counter_flag) != 2) {
+ if (embb_internal__atomic_compare_and_swap_4(
Why is the const attribute removed for some functions instead of declaring members mutable? The const attribute should reflect external / visible behavior.
	operator BaseType() const;
+ operator BaseType();
	BaseType Load() const;
+ BaseType Load();
	BaseType* operator->() const;
	BaseType& operator*() const;
+ BaseType* operator->();
+ BaseType& operator*();
Why does the C++ implementation contain any mutex-specific code at all? Ideally, this is completely handled at C level. Example:
template<typename BaseType>
inline void AtomicInteger<BaseType>::operator|=(BaseType val)
{ NativeType native_operand; memcpy(&native_operand, &val, sizeof(val)); + EMBB_ATOMIC_MUTEX_LOCK(this->internal_mutex); or_assign_implementation<NativeType>:: or_assign(&this->AtomicValue, native_operand); + EMBB_ATOMIC_MUTEX_UNLOCK(this->internal_mutex); }
Do we have any mechanism now in DEBUG mode to detect uninitialized atomics?

--------------------------------------------------

"Why are EMBB types / functions replaced with corresponding internal ones?"
These are used in places where static initializers are used. The static atomics all use the internal implementation which always works without a mutex, because there is no way to safely initialize a corresponding mutex for them.
"Why is the const attribute removed for some functions instead of declaring members mutable?"
Mutable it is then. I just wanted to make clear that these functions indeed write to the atomic.
"Why does the C++ implementation contain any mutex-specific code at all?"
The C++ implementation maps to the internal atomic implementation which has no notion of the atomic variable struct and instead directly works on the atomic value. This was simply the cleaner way to get mutexes into the C++ implementation. Also I did not want to rewrite it completely.
"Do we have any mechanism now in DEBUG mode to detect uninitialized atomics?"
Yes. See the marker member of the atomic variable struct and the EMBB_ATOMIC_* macros.

--------------------------------------------------

Made internal mutex mutable: https://github.com/siemens/embb/commit/696bc68d65fb9ec2bae14fe6716025a25f99e488

Updated and fixed documentation: https://github.com/siemens/embb/commit/61caf1abf8a9615dbdacbdd3c6020841dddd96b7

--------------------------------------------------

Initialization functions with optional parameter (initial value, default: 0)
Enable marker check in DEBUG mode

--------------------------------------------------

Done: https://github.com/siemens/embb/commit/f58e653ac76b7b0b2be6bcdd7e6c84cdc1272513
Note: default parameter values are unsupported in c, thus init parameters are mandatory.

marcus-winter, 2016-11-22:
Fixed Jenkins issues. Please validate.

marcus-winter, 2016-11-24:
Jenkins failed the tests, there seems to be a race condition in the MTAPI shutdown code :(

marcus-winter, 2016-11-25:
Applied hotfix https://github.com/siemens/embb/commit/bfddc77f329b029ed7998fa31690054c2749ce2e

marcus-winter, 2016-11-29:
Jenkins tests ran through.

closed 2016-11-29 by marcus-winter