siemens/embb #16: The tasklimit is reached in some configurations of a dataflow network

tobias-schuele, 2016-11-11:
Check whether there is an error in the determination of the maximum number of slices when constructing the MTAPI scheduler. Also, add "\note" in Doxygen documentation that tell users how

- autoconfiguration works
- task limit of scheduler may have to be increased when increasing the number of slices (as otherwise, the latter may not have any effects)

marcus-winter, 2016-11-24:
The error was fixed in https://github.com/siemens/embb/commit/64b15b2a6803fd26ceab946b2b1984ac76c6bb6b. However, another issue remains: If algorithms and dataflow are used together, both will determine their maximum task count based on the MTAPI node task limit. In this case the task limit will be exceeded. We need to discuss how to resolve this, therefore I'll create a separate issue.

marcus-winter, 2016-11-24:
I added notes as requested, see #28.

marcus-winter, 2016-11-24:
Created #30 for discussion on nesting issue mentioned above.

PR #28 (2016-11-24): dataflow_cpp: added notes on automatic configuration and reduction of slices
Fixes #16
closed 2016-11-24 by tobias-schuele