siemens/embb #17: When using ForEach in dataflow serial process, new slice is processed before completion

tobias-schuele, 2016-11-11:
When using ForEach inside a Serial Process in Dataflow, it seems like the Process is respawning the task before the ForEach has finished computation. This can lead to wrong results, as the code behind the ForEach may depend on an internal state.

Pseudocode:
SerialProcess
    DebugOutput BEFORE
    ForEach
    DebugOutput AFTER

Without ForEach, the debug outputs are printed in serial. With the ForEach, they are grouped to blocks. E.g. Before, Before, Before, Before, After, After, After, After. This may indicate that a new slice is processed when the task waits for the ForEach to finish.

tobias-schuele, 2016-11-11:
The MTAPI queue implementation has no separate storage and emits tasks to a specific worker. This works correctly as long as n queue task calls wait. If wait is called a subsequent queued task might interleave the execution of the current queued task. The fix requires a separate task queue for each MTAPI queue.

marcus-winter, 2016-11-22:
Please review #25

PR #25 (2016-11-23): Fix for issue #17
Fixes #17
closed 2016-11-23 by tobias-schuele