siemens/embb #19: Revise tutorial

tobias-schuele, 2016-11-16:
The tutorial needs to be revised and extended. As a first step, outline what has to be changed in which sections. This includes a description of those parts that have been added since the last version of the tutorial such as support for heterogeneous systems (algorithms, dataflow, CUDA plugin, etc.)

marcus-winter, 2017-01-20:
Sections for heterogeneous algorithms and dataflow need to be added. There is no tutorial for the CUDA plugin. Everything else is covered albeit a bit brief.

tobias-schuele, 2017-02-24:
Updated tutorial is available in embb19-revise_tutorial. I propose to move the section on MTAPI at the end and to start with the high-level patterns (algorithms). Changes to the text are probably necessary to maintain a consistent flow. Final review will follow once support for heterogeneous systems is integrated.

PR #78 (2017-04-07): Embb19 revise tutorial
Fixes #19
closed 2017-04-07 by tobias-schuele