siemens/embb #21: Blocking data structures

tobias-schuele, 2016-11-17:
Many users need concurrent data structures but not necessarily lock-free or wait-free ones (which are already available in EMB²). For example, if an application is not timing critical, blocking data structures might be beneficial w.r.t memory consumption and context switching. Implement thread-safe wrappers for the following data structures provided by the standard library:

- set
- map
- stack
- queue
- priority_queue

tobias-schuele, 2016-11-17:
Comments from JIRA:
--------------------------------------------------
Some of the structures of the std library, namely stacks, queues, priority_queues, rely on a SequenceContainer (e.g. std::vector) for the internal representation of the object. If all of those containers make use of dynamic memory allocation it might be necessary to implement those data structures from scratch. I have already done that for stacks and queues.
--------------------------------------------------
I agree, memory allocation is an issue here.
--------------------------------------------------
Needs further discussion and decision regarding what will be implemented and how the naming of the containers is.
--------------------------------------------------
I merged all the branches of the subtickets into embb499_blocking_data_structures. See https://github.com/siemens/embb/commit/757282b1a77c9443a61155b7db0519685b3c9431 https://github.com/siemens/embb/commit/3838dc3697e3e59007352487ffd47626832fa3df
--------------------------------------------------
Usually, a range is sufficient, however here we have a tree, e.g., commit 863eac has parents c511228 and c21ffa7 Please give a list of commits.
--------------------------------------------------
See:
https://github.com/siemens/embb/commit/757282b1a77c9443a61155b7db0519685b3c9431
https://github.com/siemens/embb/commit/173916a83655b3a08eedf3918bc11f709f8f351d
https://github.com/siemens/embb/commit/1c75c0135e8c82c9b9dda29f18fc0b0150819b96
https://github.com/siemens/embb/commit/d9a6be5da61ec98fa8ff16511f059f1ef26e56fc
https://github.com/siemens/embb/commit/0deb6bf8585add9dd2fced517db7af2fa857733a
https://github.com/siemens/embb/commit/e9e7006fda05a10697ccb2a12ffc7c1e483c4416
https://github.com/siemens/embb/commit/ad5d2ef1d9016632870bccac3b9d955c05bb3617
https://github.com/siemens/embb/commit/c21ffa72083fdad4d519d34684adcbb85ee81174
https://github.com/siemens/embb/commit/ce77c47e19761372067171a39a29e6fff8df43ee
https://github.com/siemens/embb/commit/54757fc75b5bb78f98f4f97a1562fe9e9f92e9b4
https://github.com/siemens/embb/commit/b4337ce6c973c8f10b165443b8a290e6a3376db3
https://github.com/siemens/embb/commit/90ba70e8765a7059df736c85e09edf141dc0e672
https://github.com/siemens/embb/commit/c511228aed20ba4678196fa636f6dc43d6faafc8
https://github.com/siemens/embb/commit/863eac1a831fb61362c8cc64a4786e64540d9eaa
https://github.com/siemens/embb/commit/526d902a869238ecfafef76ebdadf00bc1b8bfd9
https://github.com/siemens/embb/commit/ecd1ecdfe44d91c809855985d16f3a46000eb31c
https://github.com/siemens/embb/commit/b8e716253a94f9a5f14552b57384da987c35701d
https://github.com/siemens/embb/commit/e59cccfceb9d411cc87988fdd041a7a71851ebcc
https://github.com/siemens/embb/commit/db1716262caad3f8472ef48e99099d478373acd7
https://github.com/siemens/embb/commit/f5ca8fe42a08c8a15c30255b6cca0a500f0dfa8b
https://github.com/siemens/embb/commit/3838dc3697e3e59007352487ffd47626832fa3df

sebhub, 2017-05-19:
Maybe you have a look at Boost.Intrusive:

http://www.boost.org/doc/libs/1_64_0/doc/html/intrusive.html

Does it really make sense to simply wrap a lock around the container? Normally, you have the containers with additional state which needs also protection.