siemens/embb #30: When nesting algorithms or algorithms and dataflow, the tasklimit will be exceeded.

marcus-winter, 2016-11-24:
The task limit will be exceeded when nesting algorithms and/or dataflow the task limit will be exceeded because all of them determine their maximum task count based on the node task limit. For example two nested parallel loops will launch up to double the task limit tasks. We need to discuss how to resolve this issue.

tobias-schuele, 2017-02-15:
The task limit may indeed be exceeded in case of deeply nested constructs. However, this is generally not the case as the number of tasks created depends on the number of available cores but not on the task limit. For example, two nested loops will create approximately n*n tasks, where n is the number of cores, which is still less than the default task limit for most systems.

tobias-schuele, 2017-03-08:
Should be mentioned in documentation:
- README.md, section "Limitations and Important Notes"
- Algorithms and dataflow, where computation of task count (block size) is explained

PR #76 (2017-03-21): Embb30 nesting algorithms and dataflow
Fixes #30
closed 2017-03-21 by tobias-schuele