siemens/embb #37: AppVerifier warnings

tobias-schuele, 2016-12-14:
Microsoft's App Verifier V 10.0, which is part of the Windows SDK, reports some problems for the tests. It is not clear whether these are real or just false positives.

- If Basics -> Locks is enabled, embb_base_c_test.exe and embb_base_cpp_test.exe embb_containers_cpp_text.exe fail
- If Basics -> Handles is enabled, embb_mtapi_network_c_test.exe fails

marcus-winter, 2016-12-16:
Reproduced the problems. The embb_mtapi_network_c_test failure looks like a bug.

marcus-winter, 2016-12-20:
Warnings in embb_base_c_test and embb_base_cpp_test were due to double initialization of counters.

marcus-winter, 2016-12-20:
embb_containers_cpp_test fails due to objects with active critical sections being deleted. Looks like destructors of contained objects are not called in object pool while being deleted.

marcus-winter, 2016-12-20:
The problem in embb_mtapi_network_c_test is a false positive. App verifier complains that an invalid socked handle is passed to recv. At this point the handle was just created and returned as pending by select. Subsequently, if the exception is ignored, the socket in question works perfectly normal.

tobias-schuele, 2016-12-22:
Created separate issue #40 for object pools. Please mention commit ids for already resolved warnings so that this issue can be closed.

marcus-winter, 2016-12-22:
embb_mtapi_base_c_test:
https://github.com/siemens/embb/commit/f8b638c2030e3aaf011a7f31450e26854037d12e
https://github.com/siemens/embb/commit/fd74d8329d1577c2ee5e686763781c39f0175eb0
https://github.com/siemens/embb/commit/bc7a5079dbc774b26dcd3b97d345ffb574e5f874
embb_mtapi_base_cpp_test:
https://github.com/siemens/embb/commit/801c11b1c659af8f2c1b2c5e5efe17a678dca1a2

closed 2016-12-22 by tobias-schuele