siemens/embb #40: Destruction of object pool elements

tobias-schuele, 2016-12-22:
embb_containers_cpp_test fails when run with AppVerifier due to objects with active critical sections being deleted. Looks like destructors of contained objects are not called in destructor of object pool (cf. #37).

fuchsto, 2016-12-22:
In case the cause of this defect has not been identified in the meantime:

If I am not mistaken, in `embb::containers::ObjectPool`, objects are allocated and deallocated using memory allocation functions of the base C interface:
https://github.com/siemens/embb/blob/master/base_cpp/src/memory_allocation.cc#L43

As C-style memory management has no concept for constructors and destructors, the objects' underlying memory is freed without calling their destructors.

If possible and practicable, I would like to use this defect in a course work assignment for my students in the C++ Master's course and award the author of the best solution with a pull request that is officially reviewed at the multicore expert center at Siemens CT.

tobias-schuele, 2016-12-23:
We didn't have a closer look at this yet. However, the issue is independent of the underlying memory allocation scheme, as the objects in the pool are constructed using placement new (after the call to malloc) and should be destructed before the memory is freed (by calling the destructor). There are two ways to guarantee correct construction/desctruction: Either all elements in the pool are always constructed even if some of them are unused (hence, all elements can be destructed when the lifetime of the pool ends), or only those elements are constructed which are actually in use. For the latter, destruction of the pool requires to know which elements must be destructed. In any case, we'll investigate this and of course highly appreciate all kinds of support or feedback ;-)

PR #50 (2017-02-14): Embb40 destruction of object pool elements
Fixes #40
closed 2017-02-14 by tobias-schuele