siemens/embb #55: Tutorial application

tobias-schuele, 2017-02-15:
Review and revise video processing tutorial if necessary. Describe the application in the tutorial and include the source code in the examples directory.

marcus-winter, 2017-03-06:
The application contains several binary libraries. We need to discuss what to do with them.

tobias-schuele, 2017-03-08:
Keep ffmpeg but do not include binaries. Include reference where to obtain development packages (at least for Windows, Linux users can use libraries included in distribution).

Besides stream processing, also explain usage of parallel algorithms for filter(s) and as an add-on heterogeneous systems (OpenCL). Ideally, pipeline should have one stage (process) for each processing step.

tobias-schuele, 2017-04-05:
Add comments in CMakeLists.txt explaining the basics and things to be adapted such as paths. Use "MYPATH" or something similar instead of actual paths and comment out.

tobias-schuele, 2017-04-05:
README: Prerequisites, installation, building, running (but not details about implementation).
Tutorial: Explanation of the algorithms, used patterns, code snippets, etc.

PR #87 (2017-05-15): Embb55 tutorial application
Resolves #55
closed 2017-05-18 by tobias-schuele