siemens/embb #58: Integration of EMBB into own projects via CMake

tobias-schuele, 2017-02-15:
As a user, I do not want to set the paths, libs etc. manually that are required to use EMBB. CMake allows to easily integrate other projects into own applications. Provide suitable CMake files that enable such integration. See

- https://github.com/siemens/embb/commit/6c998fa8de02793fb4f4b776e005bfd8f9e88193
- https://github.com/siemens/embb/commit/7cd8f0ebc160026dc6e9fdf5386f382b1c890faf

marcus-winter, 2017-03-20:
So far we only have a description for that on the website. Should I put a section into the readme as well?

tobias-schuele, 2017-03-20:
Where on the web site? Should be definitely part of the README, I propose after the section "Using C".

marcus-winter, 2017-03-20:
https://embb.io/get_started.html sections "Creating and Running an Application"

marcus-winter, 2017-03-21:
Added a section to the README (see https://github.com/siemens/embb/commit/ee2c58151b85f29ee8fcc126a23a38b90df3c1e8). Shall I change the website, too?

tobias-schuele, 2017-03-21:
I will add a note on the website.

PR #77 (2017-03-21): Embb58 cmake integration
Fixes #58
closed 2017-03-21 by tobias-schuele