siemens/embb #79: Unit test for MTAPI CPP fails on older platform (can't set affinity)

tobias-schuele, 2017-03-24:
Arch: x86_64, Kernel: 3.19.0, GCC: 4.8.5. See attachment.

[mtapi_cpp_test.zip](https://github.com/siemens/embb/files/869061/mtapi_cpp_test.zip)

tobias-schuele, 2017-03-27:
(gdb) thread 1
[Switching to thread 1 (Thread 0x7ff2f998c700 (LWP 8500))]
#0  0x00007ff2f9cc9c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
56	in ../nptl/sysdeps/unix/sysv/linux/raise.c
(gdb) bt
#0  0x00007ff2f9cc9c37 in __GI_raise (sig=sig@entry=6) at ../nptl/sysdeps/unix/sysv/linux/raise.c:56
#1  0x00007ff2f9ccd028 in __GI_abort () at abort.c:89
#2  0x00007ff2fa2dabbd in __gnu_cxx::__verbose_terminate_handler() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#3  0x00007ff2fa2d8b96 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#4  0x00007ff2fa2d8be1 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#5  0x00007ff2fa2d8df8 in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x0000000000415456 in embb::mtapi::internal::CheckStatus (status=MTAPI_ERR_CORE_NUM)
    at /home/tobias/embb-0.5.0/mtapi_cpp/src/check_status.cc:59
#7  0x00000000004127ba in embb::mtapi::Affinity::Set (this=0x7ff2f998bda0, worker=1, state=true)
    at /home/tobias/embb-0.5.0/mtapi_cpp/include/embb/mtapi/affinity.h:108
#8  0x0000000000412197 in TaskTest::TestBasic (this=0x7fff6843ecd0)
    at /home/tobias/embb-0.5.0/mtapi_cpp/test/mtapi_cpp_test_task.cc:90
#9  0x00000000004138e8 in partest::internal::MemberFunction<TaskTest>::operator() (this=0x23edce0)
    at /home/tobias/embb-0.5.0/build/mtapi_cpp/../partest/include/partest/internal/function.h:137
#10 0x000000000042301d in partest::TestMethod::Run (info=0x23edd80)
    at /home/tobias/embb-0.5.0/build/partest/src/test_method.cc:96
#11 0x00000000004247a9 in partest::internal::partestthreads::ThreadClosureArg1<int (*)(partest::TestInstanceInfo*), partest::TestInstanceInfo*>::ThreadStart (arg=0x23eddf0)
    at /home/tobias/embb-0.5.0/build/partest/include/partest/internal/partestThreads/thread_closures.h:69
#12 0x000000000042498a in partest::internal::partestthreads::InternalThreadStart (internalArg=0x23ede10)
    at /home/tobias/embb-0.5.0/build/partest/src/internal/partestThreads/thread_implementation.cc:98
#13 0x00007ff2fa791184 in start_thread (arg=0x7ff2f998c700) at pthread_create.c:312
#14 0x00007ff2f9d8d37d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

(gdb) thread 2
[Switching to thread 2 (Thread 0x7ff2fabad780 (LWP 8499))]
#0  0x00007ff2fa79265b in pthread_join (threadid=140681546352384, thread_return=0x0) at pthread_join.c:92
92	pthread_join.c: No such file or directory.
(gdb) bt
#0  0x00007ff2fa79265b in pthread_join (threadid=140681546352384, thread_return=0x0) at pthread_join.c:92
#1  0x0000000000424a3d in partest::internal::partestthreads::ThreadImplementation::Join (this=0x23eddd0)
    at /home/tobias/embb-0.5.0/build/partest/src/internal/partestThreads/thread_implementation.cc:120
#2  0x0000000000424584 in partest::internal::Thread::Join (this=0x23eddd0)
    at /home/tobias/embb-0.5.0/build/partest/src/internal/thread.cc:43
#3  0x0000000000420e76 in partest::TestUnit::Run (this=0x23edc60)
    at /home/tobias/embb-0.5.0/build/partest/src/test_unit.cc:127
#4  0x00000000004234ca in partest::TestCase::Run (this=0x7fff6843ecd0)
    at /home/tobias/embb-0.5.0/build/partest/src/test_case.cc:49
#5  0x000000000041489d in PartestRunTests () at /home/tobias/embb-0.5.0/mtapi_cpp/test/main.cc:40
#6  0x000000000041478b in main (argc=1, argv=0x7fff6843ee28) at /home/tobias/embb-0.5.0/mtapi_cpp/test/main.cc:37

(gdb) thread 3
[Switching to thread 3 (Thread 0x7ff2f918b700 (LWP 8501))]
#0  0x00007ff2f9d5e2a7 in sched_yield () at ../sysdeps/unix/syscall-template.S:81
81	../sysdeps/unix/syscall-template.S: No such file or directory.
(gdb) bt
#0  0x00007ff2f9d5e2a7 in sched_yield () at ../sysdeps/unix/syscall-template.S:81
#1  0x0000000000426950 in embb_thread_yield () at /home/tobias/embb-0.5.0/base_c/src/thread.c:275
#2  0x000000000041da73 in embb_mtapi_scheduler_worker (arg=0x7ff2f4029dc0)
    at /home/tobias/embb-0.5.0/mtapi_c/src/embb_mtapi_scheduler_t.c:418
#3  0x0000000000426916 in embb_internal_thread_start (internalArg=0x7ff2f402a2a0)
    at /home/tobias/embb-0.5.0/base_c/src/thread.c:262
#4  0x00007ff2fa791184 in start_thread (arg=0x7ff2f918b700) at pthread_create.c:312
#5  0x00007ff2f9d8d37d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111

marcus-winter, 2017-03-29:
How many cores does the system have?

tobias-schuele, 2017-03-29:
One core (virtual machine). Problem disappears if number of cores in VM is increased.

PR #80 (2017-04-04): mtapi_cpp: fixed affinity test for single core machines
Fixes #79
closed 2017-04-04 by tobias-schuele