siemens/embb #95: Extend script to generate tutorial

tobias-schuele, 2017-05-18:
Generate tutorial (html, epub, pdf) from markdown when creating release files. Baking "tutorial.md" from "tutorial_raw.md" using "bake_tutorial.py" should already be included in the script (create_tarball_and_zip.sh). Additional steps:

	sed -i -e 's/<\/\?sub>/~/g' tutorial.md
	sed -i -e 's/<\/\?sup>/\^/g' tutorial.md
	pandoc tutorial.md --latex-engine=xelatex -V papersize=a4 -V geometry:margin=3cm -o tutorial.pdf
	pandoc tutorial.md --css=pandoc.css --self-contained -o tutorial.html
	cat tutorial.md tutorial.yml | tail -n +2 | pandoc --epub-stylesheet=pandoc.css -o tutorial.epub

tobias-schuele, 2017-05-19:
Baked markdown file (tutorial.md) should also be included in release packages (but not tutorial_raw.md).

tobias-schuele, 2017-05-24:
Required packages such as pandoc and xelatex are already installed on CI server.

tobias-schuele, 2017-05-25:
Please also check why EMBB Ntly 3 Release build 587 failed.

christian-kern, 2017-05-29:
Build 587 failed because of DOS Line Endings in two files:
_Detected Dos line endings in following files:
/tmp/tmp.Lk3tcSHCqQ/embb-1.0.0/doc/tutorial/application/CMakeLists.txt: ASCII text, with CRLF line terminators
/tmp/tmp.Lk3tcSHCqQ/embb-1.0.0/doc/tutorial/application/README.md: UTF-8 Unicode text, with very long lines, with CRLF line terminators
Error: The project guidelines forbid Dos line endings. Exiting._

This was corrected in branch "https://github.com/siemens/embb/tree/embb95_generate_tutorial".
This branch contains also the implementation for this ticket (generating tutorial from markdown).

closed 2017-05-29 by tobias-schuele