siemens/embb #96: Build error on ARM system (Sabreboard), Build #146 (19.05.2017 14:53:51)

tobias-schuele, 2017-05-19:
```
[ 34%] Building C object mtapi_c/CMakeFiles/embb_mtapi_c.dir/src/embb_mtapi_task_t.c.o
[ 35%] In file included from /home/ubuntu/workspace/base_cpp/include/embb/base/internal/atomic/atomic_base.h:37:0,
                 from /home/ubuntu/workspace/base_cpp/include/embb/base/atomic.h:35,
                 from /home/ubuntu/workspace/base_cpp/include/embb/base/internal/function0.h:35,
                 from /home/ubuntu/workspace/base_cpp/include/embb/base/function.h:233,
                 from /home/ubuntu/workspace/base_cpp/src/function.cc:27:
/home/ubuntu/workspace/base_cpp/include/embb/base/internal/atomic/atomic_utility.h:71:11: error: ‘EMBB_BASE_BASIC_TYPE_ATOMIC_8’ does not name a type
   typedef EMBB_BASE_BASIC_TYPE_ATOMIC_8 AtomicType;
           ^
Building CXX object base_cpp/CMakeFiles/embb_base_cpp.dir/src/time.cc.o
make[2]: *** [base_cpp/CMakeFiles/embb_base_cpp.dir/src/function.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 36%] Building C object mtapi_c/CMakeFiles/embb_mtapi_c.dir/src/embb_mtapi_job_t.c.o
make[1]: *** [base_cpp/CMakeFiles/embb_base_cpp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 37%] Building CXX object base_c/CMakeFiles/embb_base_c_test.dir/test/duration_test.cc.o
```

tobias-schuele, 2017-05-19:
@sebhub: Did you compile #90 on 32 or 64 bit system?

sebhub, 2017-05-22:
In "base_c/include/embb/base/c/internal/atomic/atomic.h" we have now:

```
#if defined EMBB_PLATFORM_ARCH_CXX11

typedef std::atomic_uint_least8_t EMBB_BASE_BASIC_TYPE_ATOMIC_1;
typedef std::atomic_uint_least16_t EMBB_BASE_BASIC_TYPE_ATOMIC_2;
typedef std::atomic_uint_least32_t EMBB_BASE_BASIC_TYPE_ATOMIC_4;
typedef std::atomic_uint_least64_t EMBB_BASE_BASIC_TYPE_ATOMIC_8;

#elif defined EMBB_PLATFORM_ARCH_C11

typedef atomic_uint_least8_t EMBB_BASE_BASIC_TYPE_ATOMIC_1;
typedef atomic_uint_least16_t EMBB_BASE_BASIC_TYPE_ATOMIC_2;
typedef atomic_uint_least32_t EMBB_BASE_BASIC_TYPE_ATOMIC_4;
typedef atomic_uint_least64_t EMBB_BASE_BASIC_TYPE_ATOMIC_8;

#else

typedef EMBB_BASE_BASIC_TYPE_SIZE_1 EMBB_BASE_BASIC_TYPE_ATOMIC_1;
typedef EMBB_BASE_BASIC_TYPE_SIZE_2 EMBB_BASE_BASIC_TYPE_ATOMIC_2;
typedef EMBB_BASE_BASIC_TYPE_SIZE_4 EMBB_BASE_BASIC_TYPE_ATOMIC_4;
#ifdef EMBB_64_BIT_ATOMIC_AVAILABLE
typedef EMBB_BASE_BASIC_TYPE_SIZE_8 EMBB_BASE_BASIC_TYPE_ATOMIC_8;
#endif

#endif
```

Should we define the EMBB_BASE_BASIC_TYPE_ATOMIC_8 unconditionally or
```
// Specialization for double-words
template<> struct AtomicTraits<8> {
  typedef EMBB_BASE_BASIC_TYPE_SIZE_8 NativeType;
  typedef EMBB_BASE_BASIC_TYPE_ATOMIC_8 AtomicType;
};
```
conditionally depending on EMBB_64_BIT_ATOMIC_AVAILABLE?

marcus-winter, 2017-05-24:
This error is gone. Verified on Wandboard and with crosscompilation.

closed 2017-05-24 by tobias-schuele