siemens/embb #105: Cuzz reports error in embb_mtapi_c_test.exe (Plugintest)

tobias-schuele, 2017-05-25:
Build steps:
cmake -G "Visual Studio 14 2015 Win64" .. -DTHREADING_ANALYSIS_MODE=ON
cmake --build . --config Debug
Cuzz Version: 10.0 (x64)

See attachment

[embb_mtapi_c_test.exe.0.dat.txt](https://github.com/siemens/embb/files/1030071/embb_mtapi_c_test.exe.0.dat.txt)

marcus-winter, 2017-05-26:
The problem in embb_mtapi_network_c_test is a false positive. App verifier complains that an invalid socked handle is passed to recv. At this point the handle was just created and returned as pending by select. Subsequently, if the exception is ignored, the socket in question works perfectly normal. See also #37.

marcus-winter, 2017-05-26:
The problem in embb_mtapi_c_test is a bug in mtapi where allocated objects in pools are not destroyed when the pool is destroyed.

marcus-winter, 2017-05-26:
The problem in embb_base_c_test is a problem with statement order. The mutex held in the notification test is destroyed on thread 0 before all other threads call unlock. See https://github.com/siemens/embb/commit/c56be95a08c163cc037d002d3a3ba88b04f77f54.

PR #106 (2017-05-26): Embb105 app verifier errors
Resolves #105
closed 2017-05-26 by tobias-schuele