siemens/embb #107: AppVerifier Results

tobias-schuele, 2017-05-29:
- [embb_dataflow_cpp_test.exe.240.dat.txt](https://github.com/siemens/embb/files/1035190/embb_dataflow_cpp_test.exe.240.dat.txt)
- [image]
- [log.txt](https://github.com/siemens/embb/files/1035195/log.txt)

marcus-winter, 2017-05-29:
Fixes for the last two issues are available here:
 - https://github.com/siemens/embb/commit/a8850cf0af9b662b6470cc46f000667813e12114
 - https://github.com/siemens/embb/commit/1d5b3991aa34d893d30e3ebad126b876536f6b4c

tobias-schuele, 2017-05-29:
Running MTAPI C tests ...
Running TaskTest ...
CHECK FAILED
Expression: MTAPI_SUCCESS == status
File      : C:\Users\mch11220\Repositories\embb-development\mtapi_c\test\embb_mt
api_test_task.cc
Method    : void __cdecl TaskTest::TryDetached(void)
Line      : 255

Test unit: mtapi task test
Method thread num: 0
Method iteration num: 0

marcus-winter, 2017-05-30:
Fix for last issue is available here: https://github.com/siemens/embb/commit/368a609c9ef7abe73fe5c8552d1a8d318a350a1c. There were some more potential problems with timeouts that are fixed with this patch.

tobias-schuele, 2017-06-07:
All tests except for dataflow were executed with concurrency fuzzing for more than 24h (each with approx. 1600 different schedules). Fix for issue with ordered processes in dataflow, which came up with #64, is pending.

marcus-winter, 2017-06-08:
Fix for dataflow problem is available here: https://github.com/siemens/embb/commit/6834c6a45b473f5a0349adb94909fe16fe3e3785.

tobias-schuele, 2017-06-13:
No more findings after running appverif for approx. 100h.

closed 2017-06-13 by tobias-schuele