siemens/embb #109: compilation fails: error: size of unnamed array is negative

DannyBoyKN, 2017-10-10:
I had pulled the latest updates and trying to rebuild:
```
$> git checkout -b embb-1.0
$> mkdir build
$> cd build
$> cmake .. -DCMAKE_CXX_COMPILER=g++ -DCMAKE_C_COMPILER=gcc
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/g++
-- Check for working CXX compiler: /usr/bin/g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building shared libs disabled (default)
   (set with command line option -DBUILD_SHARED_LIBS=ON/OFF)
-- Warnings are not treated as errors (default)
   (set with command line option -DWARNINGS_ARE_ERRORS=ON/OFF)
-- MTAPI automatic initialization enabled (default)
   (set with command line option -DUSE_AUTOMATIC_INITIALIZATION=ON/OFF)
-- Threading analysis mode disabled (default)
   (set with command line option -DTHREADING_ANALYSIS_MODE=ON/OFF)
-- Exceptions enabled (default) 
   (set with command line option -DUSE_EXCEPTIONS=ON/OFF)
-- Building OpenCL plugin disabled (default)
   (set with command line option -DBUILD_OPENCL_PLUGIN=ON/OFF)
-- Building CUDA plugin disabled (default)
   (set with command line option -DBUILD_CUDA_PLUGIN=ON/OFF)
-- Building tests enabled (default)
   (set with command line option -DBUILD_TESTS=ON/OFF)
-- Extracting partest to directoy /home/vitasuser/ENEA/embb/build/partest
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of char
-- Check size of char - done
-- Check size of short
-- Check size of short - done
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Check size of int
-- Check size of int - done
-- Check size of unsigned int
-- Check size of unsigned int - done
-- Check size of long
-- Check size of long - done
-- Check size of unsigned long
-- Check size of unsigned long - done
-- Check size of long long
-- Check size of long long - done
-- Check size of unsigned long long
-- Check size of unsigned long long - done
-- Check size of intptr_t
-- Check size of intptr_t - done
-- Check size of uintptr_t
-- Check size of uintptr_t - done
-- Check size of size_t
-- Check size of size_t - done
-- Check size of ptrdiff_t
-- Check size of ptrdiff_t - done
-- Check size of uintmax_t
[... 97 lines cut ...]
   ^
embb/base_c/src/atomic.c:44:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(size_t) != EMBB_SIZE_T_TYPE_SIZE);
   ^
embb/base_c/src/atomic.c:45:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(ptrdiff_t) != EMBB_PTRDIFF_T_TYPE_SIZE);
   ^
base_c/CMakeFiles/embb_base_c.dir/build.make:326: recipe for target 'base_c/CMakeFiles/embb_base_c.dir/src/atomic.c.o' failed
make[2]: *** [base_c/CMakeFiles/embb_base_c.dir/src/atomic.c.o] Error 1
CMakeFiles/Makefile2:181: recipe for target 'base_c/CMakeFiles/embb_base_c.dir/all' failed
make[1]: *** [base_c/CMakeFiles/embb_base_c.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
```

Ubuntu 16.04.3 LTS 64bit:
```
$ uname -a
Linux ubuntu 4.10.0-35-generic #39~16.04.1-Ubuntu SMP Wed Sep 13 09:02:42 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

```
with gcc 5.4.0 and cmake 3.5.1.

Strangely, the same steps work on another machine running the same setup (except gcc is 5.4.1) !?
It is for sure a dependency issue but I cannot figure out which ?

tobias-schuele, 2017-10-12:
For a standard build, the following should be sufficient (according to the above log, you called cmake twice for generating build files):

```
cmake ..
cmake --build .
```

However, I don't think that will solve the problem. Do you have multiple compilers installed? Could you please provide the output of cmake so that we can track down the problem? Just type

`tar cfv cmake.tar CMake*`

in the build directory and attach the archive to the issue (or send it by mail to tobias.schuele@siemens.com).

Thanks

marcus-winter, 2017-10-12:
I can't reproduce the problem, but it looks like atomic_sizes.h was not configured properly by cmake. Could you please provide the files CMakeCache.txt, CMakeFiles/CMakeError.log, CMakeFiles/CMakeOutput.log and base_c/include/embb/base/c/internal/atomic/atomic_sizes.h from your build folder?

DannyBoyKN, 2017-10-12:
No, only one gcc is installed:
```
$ sudo update-alternatives --config gcc
update-alternatives: error: no alternatives for gcc
$ gcc -dumpversion
5.4.0
```

I cleared the build folder (`rm -rf *`) and rebuild acc. your instructions:
```
cmake ..
cmake --build .

```
'`cmake.tar.gz`' is attached.

```
$ make VERBOSE=1
/usr/bin/cmake -Hembb -Bembb/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start embb/build/CMakeFiles embb/build/CMakeFiles/progress.marks
make -f CMakeFiles/Makefile2 all
make[1]: Entering directory 'embb/build'
make -f partest_build/CMakeFiles/partest.dir/build.make partest_build/CMakeFiles/partest.dir/depend
make[2]: Entering directory 'embb/build'
cd embb/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" embb embb/build/partest embb/build embb/build/partest_build embb/build/partest_build/CMakeFiles/partest.dir/DependInfo.cmake --color=
make[2]: Leaving directory 'embb/build'
make -f partest_build/CMakeFiles/partest.dir/build.make partest_build/CMakeFiles/partest.dir/build
make[2]: Entering directory 'embb/build'
make[2]: Nothing to be done for 'partest_build/CMakeFiles/partest.dir/build'.
make[2]: Leaving directory 'embb/build'
[  8%] Built target partest
make -f partest_build/CMakeFiles/partest_test.dir/build.make partest_build/CMakeFiles/partest_test.dir/depend
make[2]: Entering directory 'embb/build'
cd embb/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" embb embb/build/partest embb/build embb/build/partest_build embb/build/partest_build/CMakeFiles/partest_test.dir/DependInfo.cmake --color=
make[2]: Leaving directory 'embb/build'
make -f partest_build/CMakeFiles/partest_test.dir/build.make partest_build/CMakeFiles/partest_test.dir/build
make[2]: Entering directory 'embb/build'
make[2]: Nothing to be done for 'partest_build/CMakeFiles/partest_test.dir/build'.
make[2]: Leaving directory 'embb/build'
[ 10%] Built target partest_test
make -f base_c/CMakeFiles/embb_base_c.dir/build.make base_c/CMakeFiles/embb_base_c.dir/depend
make[2]: Entering directory 'embb/build'
cd embb/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" embb embb/base_c embb/build embb/build/base_c embb/build/base_c/CMakeFiles/embb_base_c.dir/DependInfo.cmake --color=
make[2]: Leaving directory 'embb/build'
make -f base_c/CMakeFiles/embb_base_c.dir/build.make base_c/CMakeFiles/embb_base_c.dir/build
make[2]: Entering directory 'embb/build'
[ 10%] Building C object base_c/CMakeFiles/embb_base_c.dir/src/atomic.c.o
cd embb/build/base_c && /usr/bin/cc  -D_GNU_SOURCE -Iembb/base_c/include -Iembb/build/base_c/include -Iembb/base_c/test -Iembb/build/base_c/../partest/include  -pthread -std=c99 -Wall -Wextra -O3 -DNDEBUG -DNDEBUG   -o CMakeFiles/embb_base_c.dir/src/atomic.c.o   -c embb/base_c/src/atomic.c
embb/base_c/src/atomic.c: In function ‘__embb_atomic_internal_compile_time_checks’:
embb/base_c/src/atomic.c:38:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(long) != EMBB_LONG_TYPE_SIZE);
   ^
embb/base_c/src/atomic.c:39:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(unsigned long) != EMBB_UNSIGNED_LONG_TYPE_SIZE);
   ^
embb/base_c/src/atomic.c:42:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(intptr_t) != EMBB_INTPTR_T_TYPE_SIZE);
   ^
embb/base_c/src/atomic.c:43:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(uintptr_t) != EMBB_UINTPTR_T_TYPE_SIZE);
   ^
embb/base_c/src/atomic.c:44:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(size_t) != EMBB_SIZE_T_TYPE_SIZE);
   ^
embb/base_c/src/atomic.c:45:3: error: size of unnamed array is negative
   BUILD_BUG_ON(sizeof(ptrdiff_t) != EMBB_PTRDIFF_T_TYPE_SIZE);
   ^
base_c/CMakeFiles/embb_base_c.dir/build.make:326: recipe for target 'base_c/CMakeFiles/embb_base_c.dir/src/atomic.c.o' failed
make[2]: *** [base_c/CMakeFiles/embb_base_c.dir/src/atomic.c.o] Error 1
make[2]: Leaving directory 'embb/build'
CMakeFiles/Makefile2:181: recipe for target 'base_c/CMakeFiles/embb_base_c.dir/all' failed
make[1]: *** [base_c/CMakeFiles/embb_base_c.dir/all] Error 2
make[1]: Leaving directory 'embb/build'
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2

[cmake.tar.gz](https://github.com/siemens/embb/files/1379017/cmake.tar.gz)

```

DannyBoyKN, 2017-10-12:
CMakeCache.txt, CMakeFiles/CMakeError.log, CMakeFiles/CMakeOutput.log and base_c/include/embb/base/c/internal/atomic/atomic_sizes.h attached:

[CMakeCache.txt](https://github.com/siemens/embb/files/1379020/CMakeCache.txt)
[CMakeError.log](https://github.com/siemens/embb/files/1379023/CMakeError.log)
[CMakeOutput.log](https://github.com/siemens/embb/files/1379024/CMakeOutput.log)
[atomic_sizes.h.txt](https://github.com/siemens/embb/files/1379027/atomic_sizes.h.txt)

DannyBoyKN, 2017-10-12:
ok, the version of embb on this x86 machine seemed to be corrupted somehow. I re-cloned fully again and rebuild now successfully :-)

closed 2017-10-12 by DannyBoyKN